elegos/Linux-Arctis-Manager-Legacy #1: PyQt6 and Gnome major issue

XiokroDarc, 2025-01-06:
I know this probably has nothing to do with this project. But trying to install this dependency has been a total failure on Fedora. Installing it turns the gnome theme to light mode without a choice and no way of changing it. On top of that, the settings window refuses to show up at all even with the dependency installed.

elegos, 2025-01-09:
Hello @XiokroDarc,

does the install process by itself turn the gnome theme to light mode? It install its dependencies in its own folders, which sounds strange. Also it sets no theme, but rather uses the system's. I'm not using GNOME by a while (years), though I remember there was a settings section dedicated to Qt applications, you might want to tinker with it.

As per the system tray app: it shows a contextual menu only on supported devices, which for now is the Arctis Nova Pro Wireless; otherwise the icon will show up to tell you it's "working" on its basic functionalities (in case of Arctis 7+, which I think is your case), you should see the Arctis 7+ Game and Arctis 7+ Chat channels under the sound settings / volume.

XiokroDarc, 2025-01-09:
Hi I was able to figure out which file in particular causes the gnome light mode issue. `/usr/local/lib/arctis-manager/PyQt6/Qt6/plugins/platforms/libqxcb.so`

What this file is or why it is causing the issue is beyond my knowledge, all I know is if i remove the file, dark mode returns on reboot, if i place the file back, the bug returns.

However though the icon pops up and the drop menu doesn't work (which is the function Im expecting if Im right?) the actual audio inputs are not showing up like the other repo did after I did my patch to it. I would also like to add that the icon in the taskbar is that of what would be light mode, which tells me PyQt6 is wanting to be in light mode or at least is thinking it is, or is trying to force the system to be light mode and removing the .so is stopping it.

running `journalctl --user -b -f -u arctis-manager` gives

`Jan 09 00:45:20 fedora arctis-manager[2535]:               Daemon     INFO | - Arctis Manager is starting. -
Jan 09 00:45:20 fedora arctis-manager[2535]:               Daemon     INFO | -                      v 1.5  -
Jan 09 00:45:20 fedora arctis-manager[2535]:               Daemon     INFO | -------------------------------
Jan 09 00:45:20 fedora arctis-manager[2535]:               Daemon     INFO | Registering supported devices.
Jan 09 00:45:20 fedora arctis-manager[2535]:               Daemon     INFO | Registering device "Arctis 7+"
Jan 09 00:45:20 fedora arctis-manager[2535]:               Daemon     INFO | Registering device "Arctis 7+ (Destiny)"
Jan 09 00:45:20 fedora arctis-manager[2535]:               Daemon     INFO | Registering device "Arctis 7+ (PS5)"
Jan 09 00:45:20 fedora arctis-manager[2535]:               Daemon     INFO | Registering device "Arctis 7+ (Xbox)"
Jan 09 00:45:20 fedora arctis-manager[2535]:               Daemon     INFO | Registering device "Arctis Nova Pro Wireless"
Jan 09 00:45:20 fedora arctis-manager[2535]:               Daemon     INFO | Registering device "Arctis Nova Pro Wireless X"
`

which seems to be normal.

elegos, 2025-01-09:
Hello @XiokroDarc ,

I'm going to investigate the theme issue with a virtual machine.

As per the output provided, it actually registers the device managers, but don't initialize the device itself. In my knowledge, this means another project is already bound to the device itself (for example running two instances of the application, or running both this and the original project).

Please ensure you uninstall / disable the previous service first, then try unplug / replug the device in and let me know!

XiokroDarc, 2025-01-10:
@elegos i have now tested this on a fresh install of Fedora KDE on bare metal. I am not able to get the program to launch as it just instantly crashes for me. Pip and all dependencies appear to be installed. Running it directly I notice a few things
in the output.

`SystrayApp     INFO | Starting Systray app.`
`SystrayApp CRITICAL | 'QApplication' object has no attribute 'exec_'`
`    Daemon     INFO | Received shutdown signal, shutting down.`
`    Daemon     INFO | Removing PulseAudio nodes.`
`    Daemon CRITICAL | 'InterfaceEndpoint' object is not iterable`

Dont know why QApplication is failing to start but this happens before device registration. The "InterfaceEndpoint is not iterable" appears at the very end after it registered all the devices.

elegos, 2025-01-10:
This kind of error drive me nuts.

There was a typo in the systray_app.py using `exec_` (used until pyqt5) instead of `exec()` (pyqt6+). Apparently though if you had a "dirty" installation with older libs, it just worked nonetheless (as on my dev environment).

I've pushed a fix on the `develop` branch. The Daemon error probably is due to a race condition where the graceful shutdown tries to iterate over a variable which is not yet initialized. I'll add the script:line to the debug info to understand better where it fails nonetheless.

Please pull the code and try again :)

Thanks

XiokroDarc, 2025-01-10:
@elegos the error gets replaced with `QCoreApplication::exec: The event loop is already running` which seems to be fine now, but I now get
`Daemon CRITICAL | 'InterfaceEndpoint' object is not iterable`
`Daemon     INFO | Received shutdown signal, shutting down.`
`Daemon     INFO | Removing PulseAudio nodes.`

I think this critical is actually what is causing the crash, as the crash still is existing, and only actually happens when my device gets connected, and not before.

XiokroDarc, 2025-01-10:
@elegos i was able to find the exact piece of code that is causing the crash
[[in the daemon file](https://github.com/elegos/Linux-Arctis-Manager/blob/423f32a413beac4b9f96d0e6c466461c8fd8c5f0/arctis_manager/arctis_manager_daemon.py#L200)] (url)

XiokroDarc, 2025-01-10:
@elegos I FOUND OUT THE PROBLEM!!!!!!!!!!!!!!!!!!
[Line 23 of the arctis 7+ device](
https://github.com/elegos/Linux-Arctis-Manager/blob/423f32a413beac4b9f96d0e6c466461c8fd8c5f0/arctis_manager/devices/device_arctis_7_plus.py#L23) this needs a bracket around the return value 
this 
`self.utility_guess_endpoint(7, 'in')`
should be this
`[self.utility_guess_endpoint(7, 'in')]`
it was expecting an array, but this wasn't returning an array like your nova device file was.

it is now FULLY functional.

elegos, 2025-01-10:
Thank you for reporting the fix itself! I reworked the device manager with release 1.4, I just missed to update the 7+ device manager!

Fixed in v1.5.3!

closed 2025-01-10 by elegos