elegos/Linux-Arctis-Manager-Legacy #3: Bazzite Installation

dokbanks, 2025-01-18:
Hi there!

Forgive me if this is a simple question, I am trying to install your Linux Arctis Manager and when running ./install.sh it fails with the error "Failed to enable unit: Unit arctis-manager.service does not exist"

I'm using Bazzite 41 KDE Plasma 6.2.5, I'm unsure if I need to give any more information, but if I do please let me know and I will get back to you as soon as I can! Thank you for taking the time to read my post.

elegos, 2025-01-20:
Hello dokbanks!

First of all: which device do you own? You can also try executing the service in debug mode, as follows:

```bash
# ensure the service is not running
systemctl --user stop arctis-manager
# start artctis-manager manually in verbose mode
arctis-manager -v
```

Also give me the output of the command `lsusb | grep Arctis`

Thanks :)

elegos, 2025-01-28:
@dokbanks have you resolved the issue by yourself, or do you steel need help?

Thanks

dokbanks, 2025-01-28:
@elegos hi there, I really do apologise about the lateness of my reply, I'll try and get back you as soon as I can, the comment alert happened to go into my spam folder on my email and I just happened to check.

The device I'm using is the Wired Steelseries Arctis Nova Pro with GameDAC2 Headset connected via USB on a Desktop PC, is this setup compatible? I will work on getting a result for what you've suggested shortly and see if I have any luck. 

Thank you for your time and your patience

elegos, 2025-01-28:
Hello @dokbanks. The Arctis Nova Pro wired might need to get support, if it's not the same id as the wireless one. We might be able to easily find out the addresses in case, if you want to tinker with [Wireshark on Windows as described here](https://github.com/elegos/Linux-Arctis-Manager/blob/main/docs/WIRESHARK.md) (I can give you more support in the process as needed on this ticket)

elegos, 2025-02-04:
Hello @dokbanks,

I've just released a tool which should help you tracking down the packets. You can find the (little) documentation [here](https://github.com/elegos/Linux-Arctis-Manager/blob/main/docs/USB_ANALYZER.md) and the release [here](https://github.com/elegos/Linux-Arctis-Manager/releases/tag/usb_analyzer_v1.0)

dokbanks, 2025-02-05:
Hi there! I should be able to take the time to have a look at this and I'll
get back to you when I can, thank you

elegos, 2026-01-01:
Partially: audio sinks are software, not hardware, and the kernel patch doesn't seem to include them. Also a proper user interface for all the features is still required, if not already included in the various UIs (KDE, GNOME etc).