elegos/Linux-Arctis-Manager-Legacy #5: how to make this work for immutable OS like bazzite

craigers521, 2025-02-07:
Just made the mistake of buying the arctis nova pro wireless and didnt realize chatmix required software to function.  Im running bazzite which is an immutable OS.  Is it possible to make this work for a read-only filesystem?  I did add a prefix for the install path but it looks like some other files are being copied elsewhere: 

```
Installing Arctis Manager...
Installing binaries in /home/craig/.local/bin
Installing desktop file in /home/craig/.local/share/applications/
Installing icon file in /usr/share/icons/hicolor/scalable/apps/
cp: cannot create regular file '/usr/share/icons/hicolor/scalable/apps//arctis_manager.svg': Read-only file system
Installing application data in /home/craig/.local/lib/arctis-manager
Installing python dependencies in /home/craig/.local/lib/arctis-manager
Installing udev rules.
cp: cannot create regular file '/usr/lib/udev/rules.d/91-steelseries-arctis.rules': Read-only file system
Installing and enabling systemd user service.
cp: cannot create regular file '/usr/lib/systemd/user/arctis-manager.service': Read-only file system
```

craigers521, 2025-02-08:
Hey man... hoping you can help me... i tried to force this to work by manually copying in the udev rules to /etc/udev and the service to /etc/systemd and now none of my headsets are working... i tried to reverse everything i did and no dice.  I have an old arctis 9 and an old arctis 7 that show up as 2 audio devices, a game output and a chat output, and now neither of those outputs will output anything.  I tried forcing it with pipewire and its not working.  Any advice to get my old headsets back to working would be greatly appreciated, not sure what i did here.

elegos, 2025-02-09:
I need to look into this, I've never tried immutable OS versions. Will find out a solution ASAP 👍

craigers521, 2025-02-09:
> I need to look into this, I've never tried immutable OS versions. Will find out a solution ASAP 👍

Thanks so much!  So i tried manually moving in the udev rules and moved the service into /etc/systemd but it was giving me a dependency error.  Also i tried to uninstall and somehow my pipewire got all screwed up and had to get into pwvu control and manually change settings back to defaults

elegos, 2025-02-10:
@craigers521 I've looked into Silverblue (Fedora's immutable spinoff). Reading around the documentation, I figured out you are actually forbidden to install custom components outside your home directory, essentially. That'd be a problem, if you want to install udev rules, particularly. I could figure out a "local" install, though the udev rules still need to be applied, and won't be read user-level (because, you know, you might want to get access to unauthorized hardware otherwise :D).

On Silverblue there is a way to install rpm packages via rpm-ostree, and that should apply to Bazzite too (it is a Fedora spin afterall, isn't it?).

Can you please try installing the rpm and see if that works?

The `PREFIX` variable is intended to install location-customizable files in a different directory rather than /usr/local (which is by the *Filesystem Hierarchy Standard* is already the less intrusive system prefix, for host-specific installations). There are other files that can't be installed elsewhere, like indeed udev rules, system-wide systemctl services (if you want to target multi users), desktop entries, icons (always targeting multi-users).

craigers521, 2025-02-10:
Hey @elegos - yes bazzite is based on silverblue, just comes pre-packaged with a lot of tools/utilities/software that are popular with gamers.  My bad I didnt realize you had packaged this as an rpm.  I will try to layer it in and let you know!

craigers521, 2025-02-11:
i'm gonna return this headset and buy an audeze maxwell that presents as 2 separate audio outputs and doesn't require proprietary software, like steelseries used to.  i appreciate the good faith effort tho elegos and it looks like a really cool project you have hear.  Bazzite/Silverblue just isn't the best OS to deal with hardware peripherals.

closed 2025-02-11 by craigers521

elegos, 2025-02-11:
That's strange that it doesn't recognize the rpm at all. I'll try to figure it out in any case. Good catch on that other device, will take a look myselfz too :D