elegos/Linux-Arctis-Manager-Legacy #6: Broken dependencies in pre-built rpms

krokwen-tftc, 2025-02-08:
On try to install rpm from the release:

```
$ sudo dnf install https://github.com/elegos/Linux-Arctis-Manager/releases/download/v1.6.1/arctis-manager-1.6.1-1.fc41.x86_64.rpm
Updating and loading repositories:
Repositories loaded.
 https://github.com/elegos/Linux-Arctis-Manager/releases/download/v1.6.1/arctis-manager-1.6.1-1.fc41.x86_64.rpm                                                                                                                             100% |   9.6 MiB/s |  56.3 MiB |  00m06s
Failed to resolve the transaction:
Problem: conflicting requests
  - nothing provides /home/gfurlan/.local/share/virtualenvs/Arctis-Manager-Fiz9eJqG/bin/python3 needed by arctis-manager-1.6.1-1.fc41.x86_64 from @commandline
  - nothing provides libQt63DCore.so.6(Qt_6_PRIVATE_API)(64bit) needed by arctis-manager-1.6.1-1.fc41.x86_64 from @commandline
  - nothing provides libQt63DQuickScene2D.so.6(Qt_6_PRIVATE_API)(64bit) needed by arctis-manager-1.6.1-1.fc41.x86_64 from @commandline
  - nothing provides libQt63DRender.so.6(Qt_6_PRIVATE_API)(64bit) needed by arctis-manager-1.6.1-1.fc41.x86_64 from @commandline
  - nothing provides libQt6EglFSDeviceIntegration.so.6(Qt_6_PRIVATE_API)(64bit) needed by arctis-manager-1.6.1-1.fc41.x86_64 from @commandline
  - nothing provides libQt6EglFsKmsGbmSupport.so.6(Qt_6_PRIVATE_API)(64bit) needed by arctis-manager-1.6.1-1.fc41.x86_64 from @commandline
  - nothing provides libQt6EglFsKmsSupport.so.6(Qt_6_PRIVATE_API)(64bit) needed by arctis-manager-1.6.1-1.fc41.x86_64 from @commandline
  - nothing provides libQt6QmlCompiler.so.6(Qt_6_PRIVATE_API)(64bit) needed by arctis-manager-1.6.1-1.fc41.x86_64 from @commandline
  - nothing provides libQt6WebEngineQuick.so.6(Qt_6_PRIVATE_API)(64bit) needed by arctis-manager-1.6.1-1.fc41.x86_64 from @commandline
  - nothing provides libQt6WebView.so.6(Qt_6_PRIVATE_API)(64bit) needed by arctis-manager-1.6.1-1.fc41.x86_64 from @commandline
  - nothing provides libQt6WebViewQuick.so.6(Qt_6_PRIVATE_API)(64bit) needed by arctis-manager-1.6.1-1.fc41.x86_64 from @commandline
  - nothing provides libmimerapi.so()(64bit) needed by arctis-manager-1.6.1-1.fc41.x86_64 from @commandline
  - nothing provides libmimerapi.so(MICROAPI_11.1100)(64bit) needed by arctis-manager-1.6.1-1.fc41.x86_64 from @commandline
You can try to add to command line:
  --setopt=optional_metadata_types=filelists to load additional filelists metadata
  --skip-broken to skip uninstallable packages
```

Platform: Fedora 41

krokwen-tftc, 2025-02-08:
Almost the same with manually built rpms from provided spec. Same libraries issues, but no virtualenv issue

elegos, 2025-02-09:
It seems there are some symlinks when the spec file is being built. Strange though that it still links to `/home/gfurlan` when you actually build from sources. Will investigate on this.

elegos, 2025-02-10:
Ok, I'm working on a modification of the `install.sh` script and the spec file. Apparently it's just wrong to install pip dependencies via `--target`, while it should be used a combination of `--root` and `--prefix` (root = build dir, prefix = dest dir), though some dependencies like `pylupdate6` (and possibly others) ignore the `--root` argument and try to install in `/usr/bin` nonetheless.

I'm tinkering around to think how to address this issue (thinking about chrooting?)

elegos, 2025-02-10:
@krokwen-tftc I've fixed the issue and should now work correctly with [version 1.6.2](https://github.com/elegos/Linux-Arctis-Manager/releases/tag/v1.6.2).

I've checked on a virtual machine running Fedora 41.

commit b3bef9e5 (2025-02-10): Installer file now installs bundled binaries instead of python files, to address pip pre-built binary dependencies pointing to non-existing paths on the local system. Fixes #6
commit 129ec49c (2025-02-10): Fix #6 (possibly) by reconfiguring how pip is invoked for dependencies installation
closed 2025-02-10 by elegos