elegos/Linux-Arctis-Manager-Legacy #7: Weird Pipewire Screen Capture Bug

XiokroDarc, 2025-02-08:
For some odd reason, under the very specific circumstance of logging out and logging in again breaks any pipewire screen capture source while the headset is plugged in.

Steps to reproduce (consistent in my testing)

1. Have Arctis-Manager Installed
2. Plug in headset
3. Restart PC
4. Log out
5. Log in

The Pipewire Screen Capture source for OBS and any other screen capture program that relies on pipewire just fails to work.

elegos, 2025-02-09:
The application creates a virtual sink, which delivers to the main Arctis device. I will make some attempts to reproduce the problem with OBS.

In the meanwhile, have you noticed if once relogged the audio still comes through the headphones? If not, it might rely there the problem (easiest workaround would be to replug the DAC in case)

XiokroDarc, 2025-02-09:
> In the meanwhile, have you noticed if once relogged the audio still comes through the headphones? If not, it might rely there the problem (easiest workaround would be to replug the DAC in case)

Audio still stays the same which is what I find weird. I didn't even notice the bug until I tried to screen share on Fedora one day (this is a pretty fresh install with very little modifications done outside of theming). The only way to get the screen sharing to work is if I relog into my account without the headset plugged in.

elegos, 2025-02-21:
Hello @XiokroDarc 

I tried reproducing the problem, but I can't do that.

Steps I tried to reproduce:

1. Install Arctis Manager (already installed)
2. Open OBS Studio, check mic & desktop audio to ensure they're correctly detected
3. Log-off (no restart; simply session end)
4. Log-in
5.  Open OBS Studio again, checking mic & desktop audio

Result: OBS Studio is still able to capture audio

[image]

[image]

If you could share your system info and the OBS audio capture section and in case try to explain how to reproduce the problem, I could investigate further.

Thanks :)

XiokroDarc, 2025-02-21:
> If you could share your system info and the OBS audio capture section and in case try to explain how to reproduce the problem, I could investigate further.
> 
> Thanks :)

Sorry if i failed to explained this correctly. It's not the audio capture that's failing. It's the video capture that is, specifically the desktop. And I was able to deduce that at this this program was somewhat of a cause, because uninstalling resolved the issue. (dont know if its in conflict with something else)

Operating System: Fedora Linux 41
KDE Plasma Version: 6.3.1
KDE Frameworks Version: 6.11.0
Qt Version: 6.8.2
Kernel Version: 6.12.15-200.fc41.x86_64 (64-bit)
Graphics Platform: Wayland
Processors: 32 × AMD Ryzen 9 7950X 16-Core Processor
Memory: 61.9 GiB of RAM
Graphics Processor 1: AMD Radeon RX 6800
Graphics Processor 2: AMD Radeon Graphics
Manufacturer: ASUS

And this is my system details.

elegos, 2025-02-22:
I'm sorry, you specified PipeWire, but I'm not really into screencasting :D

I set up a scene in OBS Studio and added a PipeWire capture device (my webcam, the only one that I could actually select).

I still can't reproduce the problem though, as even logging off and on again, I still see the webcam's input.

Maybe there's something to do with drivers (strange though, as the service runs at user-level...).

May you please try to reproduce the problem again and see if `sudo dmesg` shows anything useful?

XiokroDarc, 2025-02-22:
> I set up a scene in OBS Studio and added a PipeWire capture device (my webcam, the only one that I could actually select).
>
> I still can't reproduce the problem though, as even logging off and on again, I still see the webcam's input.

That is the bug, you should be able to also select your desktop as an optional video output, but it disappears with this.

No Arctis Manager
[image]

With Arctis Manager
[image]

elegos, 2025-02-22:
Hello @XiokroDarc 

I use pipewire to manage (create and eventually destroy) the nodes, so it might have something to do with pipewire itself. I've opened an issue on their side, hoping that they can help me identify and fix (either side) the issue.

https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/4571

XiokroDarc, 2025-02-28:
That's good that you were able to recreate the bug, so it's not just a my end issue. But it's also sad to hear it might be a pipewire bug. I hope they are able to find a solution on their end, because this might cause issues with other programs that might try and do something similar to this.

elegos, 2025-02-28:
I'm going to do some further testing, i.e. disabling the Daemon and trying to configure the thing manually step by step and try to figure out what exactly is messing with the pw sources. As soon as I get 30 minutes of time xD