elegos/Linux-Arctis-Manager-Legacy #8: Python SyntaxError

Kentaree, 2025-02-10:
I'm having various issues running this on Linux Mint, including the tray app but also the service itself seems to fail:

```
$ journalctl --user -b -f -u arctis-manager
Feb 10 11:04:15 desktop-linux arctis-manager[3958]:   File "/usr/local/lib/arctis-manager/./arctis_manager.py", line 18
Feb 10 11:04:15 desktop-linux arctis-manager[3958]:     os.system(f'notify-send {' '.join(args)} "{summary}" "{message.format(*args, **kwargs).replace('"', '\\"')}"')
Feb 10 11:04:15 desktop-linux arctis-manager[3958]:                                                                                                                  ^
Feb 10 11:04:15 desktop-linux arctis-manager[3958]: SyntaxError: f-string: expecting '}'
Feb 10 11:04:15 desktop-linux systemd[2522]: arctis-manager.service: Main process exited, code=exited, status=1/FAILURE
Feb 10 11:04:15 desktop-linux systemd[2522]: arctis-manager.service: Failed with result 'exit-code'.
Feb 10 11:04:16 desktop-linux systemd[2522]: arctis-manager.service: Scheduled restart job, restart counter is at 5.
Feb 10 11:04:16 desktop-linux systemd[2522]: arctis-manager.service: Start request repeated too quickly.
Feb 10 11:04:16 desktop-linux systemd[2522]: arctis-manager.service: Failed with result 'exit-code'.
Feb 10 11:04:16 desktop-linux systemd[2522]: Failed to start arctis-manager.service - Arctis Manager.

```

The version of python I'm running is 3.11.11:

```
$ python -V
Python 3.11.11

```

I've tried uninstalling and re-installing which also hasn't helped. If I need to provide any more info, please let me know

elegos, 2025-02-10:
Hello @Kentaree,

are you using an up-to-date version of Linux Mint? I'd try to install a VM and try to reproduce the error. Syntax seems to be correct though, it is strange that it complains about a missing parenthesis.

Kentaree, 2025-02-10:
Hi @elegos,

I recently updated to the very latest version:

```
$ inxi -Sxxx
System:
  Host: desktop-linux Kernel: 6.8.0-52-generic arch: x86_64 bits: 64
    compiler: gcc v: 13.3.0 clocksource: tsc
  Desktop: Cinnamon v: 6.4.6 tk: GTK v: 3.24.41 wm: Muffin v: 6.4.1 vt: 7
    dm: LightDM v: 1.30.0 Distro: Linux Mint 22.1 Xia base: Ubuntu 24.04 noble

```

I'm also confused by the syntax error, I've opened up the code in an IDE and it doesn't flag anything wrong with it. Maybe a locale issue? This is what I get as output for that:

```
LANG=en_IE.UTF-8
LANGUAGE=en_IE:en
LC_CTYPE="en_IE.UTF-8"
LC_NUMERIC="en_IE.UTF-8"
LC_TIME="en_IE.UTF-8"
LC_COLLATE="en_IE.UTF-8"
LC_MONETARY="en_IE.UTF-8"
LC_MESSAGES="en_IE.UTF-8"
LC_PAPER="en_IE.UTF-8"
LC_NAME="en_IE.UTF-8"
LC_ADDRESS="en_IE.UTF-8"
LC_TELEPHONE="en_IE.UTF-8"
LC_MEASUREMENT="en_IE.UTF-8"
LC_IDENTIFICATION="en_IE.UTF-8"
LC_ALL=

```

elegos, 2025-02-13:
Hello @Kentaree 

I've freshly installed Linux Mint 22 on a virtual machine and attached the Arctis device. It is working as expected, so I can't reproduce the problem, unfortunately :(

P.S.
The distro comes with Python 3.12, so I changed the Pipfile accordingly, and I also had to manually install `libxcb-cursor0`, which apparently is a requirement for the PyQt6 library not handled with the library itself.

I don't think it's a LOCALE issue (how could it possibly be, being a python interpreter error?)

Kentaree, 2025-02-17:
Ok, I've figured this one out. I was running Python 3.11 through pyenv, apparently 3.12 made some changes to them which are needed for the script: https://docs.python.org/3/whatsnew/3.12.html#whatsnew312-pep701

Once I changed the global version of Python to 3.12 it started working. I did have some dependency issues but that was sorted by manually installing PyQt6.

Sorry for the confusion!

elegos, 2025-02-18:
Before closing, I'm now building a binary version of the application (via PyInstaller). May you please check if the ubuntu version works on Linux Mint, too? You can see the release [here](https://github.com/elegos/Linux-Arctis-Manager/releases/tag/v1.6.2-2-rc1).

Kentaree, 2025-02-19:
The Ubuntu version installs and runs fine for me! I installed the 24.10 version as it's closest to the 24.01 base version Mint use.

closed 2025-02-20 by elegos