elegos/Linux-Arctis-Manager-Legacy #11: App won't open, service running

Melk047, 2025-02-22:
[image]

This shows when i install it, the tray icon shows, clicking it does nothing, audio is working and i can change between chat and game mixes in my audio menu. Everything is working besides the app. Will give more info if needed

Melk047, 2025-02-22:
[image]

Melk047, 2025-02-22:
app opens now from start menu, tray icon does nothing and app shows this

[image]

ubuntu 24.10 GNOME

elegos, 2025-02-22:
Hello @Melk047,

the Arctis 7+ support is very limited because I don't own the device itself and just took what the original software had to offer (i.e. volume and chatmix).

I see here in your screenshots a couple of things:

1. the channels are being created
2. the audio default sink is being redirected to Arctis 7+ Game
3. the device status needs some love for your device (it is showing all the options, not the device's subset)

Now some questions for you:

1. Can you hear through your headset?
2. Can you adjust the volume and mix the channels?
3. Can you select and use for exampel the Arctis 7+ Chat channel for Discord or similar, and hear through it?

If you reply "yes" to these previous questions, the software is operating succesfully and we need to fix the device status only.

If you know that your device supports more features (besides the equalizer, which has not been implemented yet generally speaking), you could help me improve the support for your device, if you have a Windows box and willing to execute the [script I made to help tracking the USB packages](https://github.com/elegos/Linux-Arctis-Manager/blob/main/docs/USB_ANALYZER.md)!

Melk047, 2025-02-22:
yes to all the questions, and yes i have windows installed. ill execute the script

Melk047, 2025-02-22:
The usb analyzer seems to be missing, file not found in github.

elegos, 2025-02-23:
Hello @Melk047 

the script is experimental, as I used it on my device only, though it contemplates the possibility of not having a setup step (which I think it's only for the external GameDAC devices). Indeed the script asks if you want to continue. If not, it's probably because I might not have merged the branch (strange but possible). Please try with the compiled one: https://github.com/elegos/Linux-Arctis-Manager/releases/tag/usb_analyzer_v1.0

If it still doesn't work at all, please share the CLI output

Thanks :)

Melk047, 2025-02-23:
[image]

[image]

[image]

Tried multiple different values which arent shown here, all same result though.

elegos, 2025-02-23:
At this point I think you should proceed with the "manual" way looking into Wireshark directly, as described [here](https://github.com/elegos/Linux-Arctis-Manager/blob/main/docs/WIRESHARK.md). As I never owned the Arctis 7+ it's hard for me to guide you straight forward, we have to try different paths. The main logic behind sniffing packets though is that you (1) identify your device (i.e. filtering all the USB interfaces that are not your device) and (2) identify the packets (host -> device (+ eventual replies) for settings, and device -> host (+ eventual replies) for status messages

Melk047, 2025-03-01:
windows thing would not work whatsoever BUT apparently the service actually isnt running

[image]

elegos, 2025-03-02:
I think that the guide is not clear enough, I'll try spending some time doing a video tutorial on that. 

As per the service, it runs only when a recognized arctis device is being plugged in the pc. Do you see anything in /dev/ like arctis* (`ls -l /dev/arctis*`) while the service is not running? It will be visible only if (1) it is recognized via the udev rule and (2) service is not running: it will be detached from the kernel (to gain access), so even if you stop it, it will remain detached (until the next re-plug).

If not, something wrong might be happening with udev. You could try replugging the dongle while watching dmesg (`dmesg -w`) and see if it's recognized.

elegos, 2025-03-02:
Hello again @Melk047,

I found out a problem affecting the automatic script for the devices not having an init sequence, like the Arctis 7+. I fixed it and released under the [relative releases page](https://github.com/elegos/Linux-Arctis-Manager/releases/tag/usb_analyzer_v1.1). You might want to try it and see if it works now.

As per the settings change, be sure to enter the device configuration (double click on the Engine's device line).

alialice3, 2025-03-12:
Hello, I have this same problem and I own this headset. If you wanted me to provide some information using your .exe to get it running, I'm willing to. Thanks!

elegos, 2025-03-12:
Hello @alialice3 

First of all, which OS / DE are you using? Have you tried installing the service via the rpm / deb packages in the releases section? If not, please run the install script to uninstall first, then install it via the package.

In this thread there were two different problems: working (double audio channel with wrong status) and not working (no device recognized). Which of the two are you experiencing? The first one "just works" and the UI needs some love, in the second case have you tried reconnecting the device?

Last but not least: if you know that there are some configurations that can be done in the settings, you can execute the latest sniffer for windows (see last comment), execute it and follow the instructions (after installing Wireshark with USBpcap support and restarting the machine).

alialice3, 2025-03-12:
```
Operating System: Arch Linux 
KDE Plasma Version: 6.3.2
KDE Frameworks Version: 6.11.0
Qt Version: 6.8.2
Kernel Version: 6.13.6-zen1-1-zen (64-bit)
Graphics Platform: Wayland
Processors: 16 × AMD Ryzen 7 9700X 8-Core Processor
Memory: 31.0 GiB of RAM
Graphics Processor: AMD Radeon RX 7900 XTX
Manufacturer: Gigabyte Technology Co., Ltd.
Product Name: X870 AORUS ELITE WIFI7 ICE
System Version: Default string-CF-WCP-ADO
```
Everything appears to work except the app, which shows no information about the headset.  I've tried plugging it back in. Sadly I don't think I can install RPM/deb packages without converting it and there may be issues with dependencies in doing so. You could always upload an Arch friendly version of the latest RPM/deb to the AUR if you're willing, many AUR helpers can install it easily that way.

elegos, 2025-03-12:
Ok if it works, the info is "right" to not appear corectly, it's something I've tested on the Arctis Nova Pro Wireless headset only and I'm aware of the bug. Arctis 7+ support is limited to the volume and dual channels only.

I need the list of features of the headset and the relative packets when you set them, if you want a better support.

I also wanted to try a yaml-driven device configuration, rather than a hard-coded one: that'd allow people to experiment with packets and making it easier for people to implement support for other devices, too.