elegos/Linux-Arctis-Manager-Legacy #17: Every time I boot up my computer, Arctis Manager does not detect the headset unless I reconnect the device

tormodAase, 2025-05-06:
OS Version: Linux Mint 22.1
Headset: Arctis Nova Pro Wireless

tormodAase, 2025-05-06:
On startup, Instead of reconnecting the headset, I can also manually run `systemctl --user start arctis-manager.service`, and everything will start working

I'm not very familiar with linux so I'm not sure why the service does not auto-start 😆
Maybe something went wrong during the installation?

Guarde, 2025-05-07:
Use `systemctl --user status arctis-manager` to check if the service is enabled.
If it isn't, enable it via `systemctl --user enable arctis-manager`

tormodAase, 2025-05-07:
Hi @Guarde, thanks for your input!

When I boot up my system, and run `systemctl --user status arctis-manager`, I get this status:
```
arctis-manager.service - Arctis Manager
     Loaded: loaded (/usr/lib/systemd/user/arctis-manager.service; enabled; preset: enabled)
     Active: inactive (dead)
```

When enabling the service, I get this warning:
```
tormod@tormod-mint:~$ systemctl --user enable arctis-manager
Unit /usr/lib/systemd/user/arctis-manager.service is added as a dependency to a non-existent unit dev-steelseries-arctis.device.

```

I'll investigate why I get that warning

tormodAase, 2025-08-11:
Just a quick update.

I don't know what happened, but I came home one night a few months ago and booted up my PC, and everything suddenly worked. I don't know what changed, but either way I was a happy user of this service after that :)

Closing this issue