elegos/Linux-Arctis-Manager-Legacy #19: Bazzite and immutable distro support

Eliasin, 2025-05-10:
Hello!

I'm also using bazzite like #5 and both the `install.sh` and `rpm` don't work out the box. I did have success installing using the script after some modifications. Since `/usr` is a readonly filesystem when using `rpm-ostree`, I had to change most of the install directories.

```sh
# Install directories
applications_dir="${HOME}/.local/share/applications/"
icons_dir="${HOME}/.local/share/icons/hicolor/scalable/apps/"
bin_dir="${HOME}.local/bin"
udev_dir="/etc/udev/rules.d/"
systemd_dir="${HOME}/.local/share/systemd/user/"
```

Alternatively, I found out that the reason why the user in #5 said "the rpm wasn't recognized" wasn't that the rpm was formatted incorrectly, but because `rpm-ostree` gives a bad error message, the `Checkout arctis-manager-1.6.3-1.fc41.x86_64: opendir(local): No such file or directory` message refers to being unable to open `local` in `/usr` because `/usr/local` is symlinked to `/var/usrlocal` mentioned [here](https://ostreedev.github.io/ostree/adapting-existing/). This seems to be because `/usr/local` is supposed to be untouched by package managers traditionally, so `rpm-ostree` goes to great lengths to not modify it. I changed the packaging scripts to target `/usr/` instead of `/usr/local/` and `rpm-ostree install` worked perfectly (I'll have a PR up in a minute, I also ran into a few small issues I want to ask about).

I haven't been able to do any functionality tests because I own a Arctis Nova 7 which isn't supported yet but I want to try to add that too.

MrAztecL, 2025-05-15:
I've followed the instructions, but when trying to launch the application, the notification pops up:

> Arctis Manager service is not available. Please connect the device first and try opening the app again.

Last few lines of installation also don't seem happy:
```
Created symlink '/home/mraztecl/.config/systemd/user/dev-steelseries-arctis.device.wants/arctis-manager.service' → '/home/mraztecl/.local/share/systemd/user/arctis-manager.service'.
Unit /home/mraztecl/.local/share/systemd/user/arctis-manager.service is added as a dependency to a non-existent unit dev-steelseries-arctis.device.
Created symlink '/home/mraztecl/.config/systemd/user/graphical-session.target.wants/arctis-manager.service' → '/home/mraztecl/.local/share/systemd/user/arctis-manager.service'.
Job for arctis-manager.service failed because the control process exited with error code.
See "systemctl --user status arctis-manager.service" and "journalctl --user -xeu arctis-manager.service" for details.
```

TyanColte, 2025-05-16:
Running Bazzite also. On a different project with a similar install method I changed the service install directory to `/etc/systemd/user/..` which is not part of the immutable section of the distro.

and the binary copy to `${HOME}/.local/bin/...`

and that worked great.

Rhodderz, 2025-11-23:
I was able to get this installed and working by running 
```
rpm-ostree usroverlay --hotfix
sudo yum install libffi-devel
curl -fsSL https://pyenv.run | bash

then finally (with the receiver plugged in, on and on the correct usb port)

PREFIX=/usr/local/share ./install.sh
```

Need to use --hotfix to make sure changes are kept across reboots.