elegos/Linux-Arctis-Manager-Legacy #22: qt6 has lib instead of lib64

tonitch, 2025-06-18:
I'm on archlinux. I'm maintaining the [aur](https://aur.archlinux.org/packages/linux-arctis-manager) package for this project.

The last version was 1.6.2 and worked. But I can't upgrade to 1.6.3
The reason is that it's trying to access lib64 but on my pc I only have lib on this line

https://github.com/elegos/Linux-Arctis-Manager/blob/4e17300f047ed502cbeaf9085227ccb7193d5878/arctis-manager.spec#L16

I don't really know what's expected here. If I can safely change it to be lib? In what case will it be lib64 instead of lib and if lib wouldn't work, how am I supposed to get the lib64 version.

I don't really use those libraries so I would appreciate your input ^^

elegos, 2025-06-30:
That path is the one of the pipenv's pyqt6 site package, which is required to be embedded in the application to avoid external dependencies. I made this decision to reduce the external dependencies to 0 and ensure the compatibility cross distributions.

If you prefere, you could comment it and remove from the analysis and add the pyqt6 package as dependency.

yarete03, 2025-09-17:
Okey, sorry for writing multiple times, i just found Archlinux probably won't be the only distro having problems with this:

Someone already made a PR because he was having the same problem on Fedora ->  https://github.com/elegos/Linux-Arctis-Manager/pull/29/commits/5879e967e9c72523af6a37784ad8568b00a81524

He just implemented the same change i made, but i think it would be great that pyqt6 dependencies would be searched on both paths to avoid any problems with any distro (again, i do not know if it is possible or not).